ska-sa/katdal #78: MStoh5

radiosky-abhik, 2017-03-09:
HI Ludwig,
                 I find a script which can convert the .h5 files to MS to do some processing on a MeerKat data.  I now need some script which can convert back from MS to .h5 files. Is there some script like this?

Cheers,
 Abhik

ludwigschwardt, 2017-03-09:
Hi Abhik,

Unfortunately this is currently a one-way process...

Regards,

Ludwig

radiosky-abhik, 2017-03-10:
Hi Ludwig,
                 Thanks. Are you working with (or you know some people in
SKA office who are) MeerKat single dish data ? I have just started to look
into few data sets (1472018997.h5, 1474580092.h5, 1474580580.h5) few days
ago, I am mainly interested in initially to study the noise properties and
the calibration issue for future Intensity mapping problem.
                 Is there some automatic flagging software available to
flag the TOD data (Time Vs Frequency) for MeerKat autocorrelation? Thanks
for your help.

Cheers,
 Abhik

--  Abhik

<abhik@phy.iitkgp.ernet.in>

ludwigschwardt, 2017-03-13:
Yes I am familiar with the single-dish data.

We have a script (katsdpscripts/RTS/RFI_Report/rfi_report.py I think) that
can flag autocorrs (although it is tuned for crosscorrs). Ask @mauch about
it.

AlecThomson, 2022-07-05:
Hi @ludwigschwardt,

Just jumping on this thread - is there any scope for producing a `mstoh5` script? This would be particularly useful for integrating with @mauch 's https://github.com/mauch/KATObitPipe pipeline.

ludwigschwardt, 2022-07-09:
That's an interesting idea! The main challenge is to provide a ChunkStore interface to CASA tables (might be easier in MSv3). Parallel reads would be nice but not absolutely necessary since you are typically working off a directory on a single disk and the disk read speed would be the limiting factor.

If you still have the original RDB file lying around, that could serve up the metadata in the usual way.