ska-sa/katdal #81: Splitting an MS created with h5toms fails

bennahugo, 2017-04-07:
Failing to split out a calibrated field from duel correlation data in CASA 4.7. The error isn't particularly informative, so not sure what triggers this. Anyone has any ideas of what CATEGORY refers to? The closest thing I can find in memo 229 is the FLAG_CATEGORY field, but that looks fine to me. Even more puzzling things work fine on single correlation data taken earlier:
```
vis                 = 'msdir/1491291289.ms' #  Name of input Measurement set or Multi-MS
outputvis           = 'msdir/1491291289.deep2.ms' #  Name of output Measurement set or Multi-MS
keepmms             =       True        #  If the input is a Multi-MS the output will also be a Multi-MS.
field               = 'DEEP_2'        #  Select field using ID(s) or name(s).
spw                 =         ''        #  Select spectral window/channels.
scan                =         ''        #  Select data by scan numbers.
antenna             =         ''        #  Select data based on antenna/baseline.
correlation         =         'XX'        #  Correlation: '' ==> all, correlation='XX,YY'.
timerange           =         ''        #  Select data by time range.
intent              =         ''        #  Select data by scan intent.
array               =         ''        #  Select (sub)array(s) by array ID number.
uvrange             =         ''        #  Select data by baseline length.
observation         =         ''        #  Select by observation ID(s).
feed                =         ''        #  Multi-feed numbers: Not yet implemented.
datacolumn          = 'corrected'       #  Which data column(s) to process.
keepflags           =       True        #  Keep *completely flagged rows* instead of dropping them.
width               =          1        #  Number of channels to average to form one output channel
timebin             =       '0s'        #  Bin width for time averaging

CASA <20>: go split
---------> go(split)
Executing:  split()

2017-04-07 04:09:19     SEVERE  MSTransformDataHandler::makeMSBasicStructure    Exception filling the sub-tables: TableRecordRep::get_pointer - incorrect data type used for field CATEGORY
2017-04-07 04:09:19     SEVERE  MSTransformDataHandler::makeMSBasicStructure+   Stack Trace: 
2017-04-07 04:09:19     SEVERE  MSTransformDataHandler::makeMSBasicStructure    msdir/1491291289.deep2.ms left unfinished.
2017-04-07 04:09:19     SEVERE  mstransformer:: (file /var/rpmbuild/BUILD/casa-prerelease/casa-prerelease-4.7.0/gcwrap/tools/mstransformer/mstransformer_cmpt.cc, line 30)    Exception Reported: Error creating output MS structure
2017-04-07 04:09:19     SEVERE  split::::       Error creating output MS structure
```

bennahugo, 2017-04-07:
The oldsplit task does exactly the same.

bennahugo, 2017-04-07:
Retried with h5toms prior to @mauch 's single correlation fixes and @sjperkins 's fixed width column fixes. Works there.

bennahugo, 2017-04-07:
Note however that I get warnings:
```
2017-04-07 06:01:43     WARN    MSTransformDataHandler::copySource      Invalid SOURCE ID in SOURCE table row 1
2017-04-07 06:01:43     WARN    MSTransformDataHandler::copySource      Invalid SOURCE ID in SOURCE table row 2
2017-04-07 06:01:43     WARN    MSTransformDataHandler::copySource      Invalid SOURCE ID in SOURCE table row 3
```

mauch, 2017-04-07:
So it turns out the changes in PR79 were not as benign as I first thought. The issue here should hopefully be solved by [PR82](https://github.com/ska-sa/katdal/pull/82)

bennahugo, 2017-06-13:
Solved mark as closed

closed 2017-06-13 by bennahugo