ska-sa/katdal #86: h5toms fails to extract some scans

bennahugo, 2017-06-13:
This may be a problem with the h5 file itself - fresh installation of master into a venv:
```
(katdal)> $ which h5toms.py                                                                                                          
/home/hugo/DATASETS/meerkat_reductions/oms_ghosts/katdal/bin/h5toms.py
(katdal)> $ h5toms.py -b /home/hugo/DATASETS/meerkat_reductions/deep_2_wednesday/msdir/blank.ms -o 1497034809.ms -p HH,VV 1497034809.h5
Using 'pyrap' casacore binding to produce MS
WARNING:katdal.h5datav3:Irregular timestamps detected in file '1497034809.h5': expected 516.875 dumps based on dump period and start/end times, got 517 instead
WARNING:katpoint.catalogue:Skipped '1934-638' [radec] (already in catalogue)
WARNING:katpoint.catalogue:Skipped 'main_target' [radec] (already in catalogue)
WARNING:katpoint.catalogue:Skipped '1934-638' [radec] (already in catalogue)
WARNING:katpoint.catalogue:Skipped 'main_target' [radec] (already in catalogue)
Extract MS for spw 0: central frequency 1284.00 MHz
Will create MS output in 1497034809.ms

#### Producing MS with HH,VV polarisation(s) ####

Using m057 as the reference antenna. All targets and activity detection will be based on this antenna.

Writing static meta data...

Iterating through scans in file(s)...

scan   1 (  36 samples) loaded. Target: '1934-638'. Writing to disk...
Wrote scan data (497.250000 MB) in 7.684730 s (64.706242 MBps)

scan   3 (  37 samples) loaded. Target: 'main_target'. Writing to disk...
Wrote scan data (511.062500 MB) in 7.806931 s (65.462666 MBps)

scan   4 (  37 samples) loaded. Target: 'main_target'. Writing to disk...
Wrote scan data (511.062500 MB) in 7.825770 s (65.305074 MBps)

scan   5 (  38 samples) loaded. Target: 'main_target'. Writing to disk...
Wrote scan data (524.875000 MB) in 8.144289 s (64.447001 MBps)

scan   6 (  37 samples) loaded. Target: 'main_target'. Writing to disk...
Wrote scan data (511.062500 MB) in 8.626223 s (59.245223 MBps)

scan   7 (  38 samples) loaded. Target: 'main_target'. Writing to disk...
Wrote scan data (524.875000 MB) in 8.000473 s (65.605496 MBps)

Traceback (most recent call last):
  File "/home/hugo/DATASETS/meerkat_reductions/oms_ghosts/katdal/bin/h5toms.py", line 344, in <module>
    for scan_ind, scan_state, target in h5.scans():
  File "/home/hugo/DATASETS/meerkat_reductions/oms_ghosts/katdal/local/lib/python2.7/site-packages/katdal/dataset.py", line 873, in scans
    target = self.catalogue.targets[self.target_indices[0]]
IndexError: list index out of range
```

bennahugo, 2017-06-13:
@spassmoor Not sure if this is an observational issue - maybe the last scan(s) weren't dumped correctly or the metadata is wrong?

ludwigschwardt, 2017-06-13:
Oops, that's a katpoint error... Please update your katpoint if using the GitHub repo. I still have to make a new PyPI release.

bennahugo, 2017-06-13:
Thanks @ludwigschwardt

closed 2017-06-13 by bennahugo