ska-sa/katdal #99: add a WEIGHT_SPECTRUM column

SpheMakh, 2017-10-19:
Not having these weights may be limiting our dynamic range a bit. It is briefly mentioned in this PR https://github.com/ska-sa/katdal/pull/62, but seems to have fallen through the cracks.

SpheMakh, 2017-11-08:
I now have a tool to calculate the weights using the [spec. sensitivity stats](http://public.ska.ac.za/meerkat/meerkat-schedule), but adding the column to the MS is a bit involved. So we just need it to exist in the MS, along with a SIGMA_SPECTRUM column.

SpheMakh, 2017-11-08:
Fixed, I'll use our [msutils](https://github.com/SpheMakh/msutils/issues/5) to create and populate these columns.

SpheMakh, 2017-11-23:
Reopening based on discussion at thermal noise breakout session 

[image]

ludwigschwardt, 2019-04-05:
I'm back too :smile:

This has now been implemented in #231 and should be in the next release (v0.13). Please test!

SpheMakh, 2019-04-05:
Awesome!

SpheMakh, 2019-04-05:
Good things come to those who weight!

ludwigschwardt, 2019-05-06:
Can I close this?

SpheMakh, 2019-05-06:
Yep

closed 2019-05-06 by SpheMakh