ska-sa/katdal #116: Herd-based sensor readings

bennahugo, 2018-01-24:
@spassmoor 
Currently h5toms simply uses the first antenna by default as a "reference antenna" for sensor readings, in particular to work out scan boundaries for one:
https://github.com/ska-sa/katdal/blob/master/katdal/h5datav3.py#L401
and
https://github.com/ska-sa/katdal/blob/master/katdal/h5datav3.py#L538

This fails for datasets where the first antenna isn't tracking properly for some reason. Is there a particular reason why this reading cannot be herd-based such that when, say, more than half of the antennae are on target the state go into tracking and the rest are quack-flagged until such time that sensors indicate they reach target?

See for instance this dataset:
```
If you want a  test case, here's the culprit: 1508095451.h5

running: h5toms.py --blank-ms /var/kat/static/blank.ms --output-ms /home/sharmila/msdir/1508095451.ms --tar  --verbose  --channel-range  16997,22738 --model-data  /input/1508095451.h5
WARNING:katdal.dataset:The selection criteria resulted in an empty data set
WARNING:katdal.dataset:The selection criteria resulted in an empty data set
WARNING:katdal.dataset:The selection criteria resulted in an empty data set
Using 'pyrap' casacore binding to produce MS
Extract MS for spw 0: central frequency 1284.00 MHz
Will create MS output in /home/sharmila/msdir/1508095451.ms

#### Producing MS with HH,VV polarisation(s) ####

Channel range 16997 through 22738.

Using m045 as the reference antenna. All targets and activity detection will be based on this antenna.

Iterating through scans in file(s)...

Successful read/write open of default-locked table /home/sharmila/msdir/1508095451.ms: 25 columns, 0 rows
Writing static meta data...
Table FEED:
  opened successfully
  added 16 rows
  wrote column 'NUM_RECEPTORS' with shape (16,)
  wrote column 'SPECTRAL_WINDOW_ID' with shape (16,)
  wrote column 'RECEPTOR_ANGLE' with shape (16, 2)
  wrote column 'INTERVAL' with shape (16,)
  wrote column 'POL_RESPONSE' with shape (16, 2, 2)
  wrote column 'TIME' with shape (16,)
  wrote column 'POLARIZATION_TYPE' with shape (16, 2)
  wrote column 'FEED_ID' with shape (16,)
  wrote column 'POSITION' with shape (16, 3)
  wrote column 'BEAM_ID' with shape (16,)
  wrote column 'ANTENNA_ID' with shape (16,)
  wrote column 'BEAM_OFFSET' with shape (16, 2, 2)
  closed successfully
Table POLARIZATION:
  opened successfully
  added 1 rows
  wrote column 'CORR_TYPE' with shape (1, 2)
  wrote column 'CORR_PRODUCT' with shape (1, 2, 2)
  wrote column 'NUM_CORR' with shape (1,)
  wrote column 'FLAG_ROW' with shape (1,)
  closed successfully
Table DATA_DESCRIPTION:
  opened successfully
  added 1 rows
  wrote column 'SPECTRAL_WINDOW_ID' with shape (1,)
  wrote column 'POLARIZATION_ID' with shape (1,)
  wrote column 'FLAG_ROW' with shape (1,)
  closed successfully
Table OBSERVATION:
  opened successfully
  added 1 rows
  wrote column 'TELESCOPE_NAME' with shape (1,)
  wrote column 'LOG' with shape (1, 1)
  wrote column 'OBSERVER' with shape (1,)
  wrote column 'SCHEDULE' with shape (1, 1)
  wrote column 'RELEASE_DATE' with shape (1,)
  wrote column 'SCHEDULE_TYPE' with shape (1,)
  wrote column 'PROJECT' with shape (1,)
  wrote column 'TIME_RANGE' with shape (1, 2)
  wrote column 'FLAG_ROW' with shape (1,)
  closed successfully
Table ANTENNA:
  opened successfully
  added 16 rows
  wrote column 'NAME' with shape (16,)
  wrote column 'MOUNT' with shape (16,)
  wrote column 'DISH_DIAMETER' with shape (16,)
  wrote column 'STATION' with shape (16,)
  wrote column 'OFFSET' with shape (16, 3)
  wrote column 'POSITION' with shape (16, 3)
  wrote column 'TYPE' with shape (16,)
  wrote column 'FLAG_ROW' with shape (16,)
  closed successfully
Traceback (most recent call last):
  File "/usr/local/bin/h5toms.py", line 548, in <module>
    raise RuntimeError("No usable data found in HDF5 file "
RuntimeError: No usable data found in HDF5 file (pick another reference antenna, maybe?)
```

ludwigschwardt, 2018-01-24:
I like your terminology :-)

While this is a good workaround, the real solution we are working towards is a *prescriptive* sensor as opposed to a *descriptive* one. We want to know what the telescope is supposed to do, and don't care what it's actually doing... This will introduce a subarray-level activity sensor (`obs_activity`, most likely). Any antenna that disagrees with the plan can then be flagged as required.

The problem with the herd approach is that some experiments have the dishes legitimately doing different things - think holography.

spassmoor, 2018-01-24:
Yes . I idea being that the data is split into scans based on what the telescope was meant to do. but then the data is flagged based on what each telescope was doing. This should not be to hard to do since we already have activity flagging per antenna / per baseline .

bennahugo, 2018-01-24:
@ludwigschwardt yup that sounds even better :)

ludwigschwardt, 2018-04-24:
Now that we have an `obs_activity` sensor I'm going to base the scan boundaries on that instead. The dishes are already "quack-flagged" when they are not on target.

ludwigschwardt, 2019-02-12:
Please test the latest katdal 0.12 and let me know if that solves your problem :-)

ludwigschwardt, 2019-04-05:
Can we close this?

closed 2019-10-01 by ludwigschwardt