ska-sa/katdal #135: Missing AUTOCORRS

bennahugo, 2018-04-13:
@ludwigschwardt 
Running katdal 0.9.5 to convert some recent skarab data to MSs for processing. I use autocorrelations monitor power levels across all fields and preflag suspect data before doing calibration.

When running h5toms without -a no autocorrelations are dumped:
```
In [2]: t = tbl("/scratch/bhugo/modelling0408-65/msdir/raw/1523187060.ms")
Successful readonly open of default-locked table /scratch/bhugo/modelling0408-65/msdir/raw/1523187060.ms: 23 columns, 203532 rows

In [3]: a = t.getcol("ANTENNA1")
In [4]: b = t.getcol("ANTENNA2")
In [5]: c = t.getcol("DATA")
In [6]: c[a == b]
Out[6]: array([], shape=(0, 4096, 4), dtype=complex64)
```

Same thing happens on an older dataset 1477074305.h5

Funny enough when you do specify -a the autocorrelations are added to the MS and filled with zeros!

bennahugo, 2018-05-08:
@bmerry confirmed this is fixed on your branch:
```
CASA <10>: d["data"].T[d["antenna1"] == d["antenna2"]]
  Out[10]: 
array([[[  4.83576758e+03 +1.37450051e-23j,
           3.17733569e+03 -1.00841885e+01j,
           3.17733569e+03 +1.00841885e+01j,
           2.19321777e+03 +2.82733842e-24j],
        [  7.85354004e+01 +0.00000000e+00j,
           8.86492908e-01 -1.70606390e-01j,
           8.86492908e-01 +1.70606390e-01j,
           7.01951294e+01 +0.00000000e+00j],
```

bmerry, 2018-05-08:
That branch has already been merged, so I'm going to close.

closed 2018-05-08 by bmerry