ska-sa/katdal #143: Missing dependencies on master

bennahugo, 2018-05-03:
Following the authenticated guide. The following dependencies are missing:
```
rdbtools
botocore
```

ludwigschwardt, 2018-05-03:
See PR #139 :-) You might also want to install `python-lzf`.

ludwigschwardt, 2018-05-03:
Actually, PR #139 still needs to get updated requirements - let's leave that for the next PR. Also, you probably want to install katdal as:

```pip install katdal[ms,s3]```

since S3 and MS support is effectively optional.

bennahugo, 2018-05-08:
thanks @ludwigschwardt

bmerry, 2018-09-20:
@bennahugo is this resolved now?

ludwigschwardt, 2019-02-12:
@bennahugo, another ping now that we are on katdal 0.12...

bennahugo, 2019-02-13:
Sorry missed this one when I was on sick leave. It looks like it has been fixed a long time ago, thanks. Closing

closed 2019-02-13 by bennahugo