ska-sa/katdal #146: SPECTRAL_WINDOW subtable is missing with new RDB format conversion

bennahugo, 2018-05-08:
hugo@stevie /scratch/bhugo/G330/msdir/raw                                                                                                                                                                                                                           [19:03:34]
> $ aoflagger -strategy ../../../NGCxxx/input/jan2018_firstpass.rfis 1524852817.ms
AOFlagger 2.9.0 (2016-12-20) command line application
This program will execute an RFI strategy as can be created with the RFI gui
and executes it on one or several observations.

Author: André Offringa (offringa@gmail.com)

Modified single-baseline strategy so it will execute strategy on all baselines and write flags.
Starting strategy on 2018-May-07 19:03:47.675879
0% : strategy...
0% : +-For each measurement set...
0% : +-+-Processing measurement set 1524852817.ms...
An exception occured during execution of the strategy!
Your set might not be fully flagged. Exception was:
SSMIndex::getIndex - access to non-existing row 0 in column NUM_CHAN of table /scratch/bhugo/G330/msdir/raw/1524852817.ms/SPECTRAL_WINDOW
No polarization statistics were collected.

ludwigschwardt, 2018-05-08:
Was it there with HDF5 conversion?

bennahugo, 2018-05-08:
@ludwigschwardt there is no HDF5 in the archive for this observation - it is quite large.  I'm see @bmerry  merged in some of the very latest fixes, so I will retry with the latest master. I will let you know by end of day if it converted successfully.

bmerry, 2018-05-08:
I don't think I've touched the code that generates those subtables. I can take a look, but it will be a while because I'm on a critical path to get all the SDP infrastructure upgraded during the current downtime window.

bennahugo, 2018-05-08:
No problem @mauch successfully converted to uvfits - will convert it to uvfits and convert it to MS with CASA.

ludwigschwardt, 2018-05-08:
... and that way you also get valid weights :-)

bmerry, 2018-05-08:
I've just checked on a small observation and I'm getting a SPECTRAL_WINDOW subtable. Can you point me at the specific observation you're trying to convert?

bennahugo, 2018-05-08:
@bmerry http://stgr1.sdp.mkat.chpc.kat.ac.za:7480/1524852817/1524852817_sdp_l0.full.rdb

bmerry, 2018-05-08:
I didn't want to wait forever to convert the file so I stuck in a break at the end of the scan loop, but I seem to get a SPECTRAL_WINDOW subtable from that input.

What command-line are you using?

bennahugo, 2018-05-08:
@bmerry I asked for full correlation data:
```
mvftoms.py -f -o msdir/raw/1524852817.ms ./1524852817_sdp_l0.full.rdb
```

bmerry, 2018-05-08:
Sorry, still haven't been able to reproduce with the same flags (although again, I'm breaking out of the conversion loop early to run in a sane amount of time). I'm running v0.9.5-80-g9f74805.

bennahugo, 2018-05-09:
Sorry for the wild goose chase I must have had a slightly broken katdal last week Wednesday when I started my conversion. The latest master works.

closed 2018-05-09 by bennahugo