ska-sa/katdal #184: missing source.timestamps in 1535789408_sdp_l0.full.rdb

SpheMakh, 2018-09-06:
I'm getting this error when converting this data set into an MS using mvftoms.py. 

```
  File "/usr/local/bin/mvftoms.py", line 777, in <module>
    main()
  File "/usr/local/bin/mvftoms.py", line 254, in main
    dataset = katdal.open(open_args, ref_ant=options.ref_ant)
  File "/usr/local/lib/python2.7/dist-packages/katdal/__init__.py", line 336, in open
    ref_ant, time_offset, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/katdal/visdatav4.py", line 112, in __init__
    num_dumps = len(source.timestamps)
TypeError: object of type 'NoneType' has no len()
```

ludwigschwardt, 2018-09-07:
I can't reproduce this with the latest katdal master. The following succeeds on imgr-com-4:

```
 f = katdal.open('1535789408_sdp_l0.full.rdb')
```

I'd still like to know what fixes this since @vasaantk had this problem too.

SpheMakh, 2018-09-07:
I switched to master and that problem went away. But now I'm getting different errors:

1. The DaskLazyIndexer 

```
scan   1 (   3 samples) loaded. Target: 'J124906-1941'. Writing to disk...
Traceback (most recent call last):
  File "/usr/local/bin/mvftoms.py", line 791, in <module>
    main()
  File "/usr/local/bin/mvftoms.py", line 541, in main
    scan_vis_data, scan_weight_data, scan_flag_data)
  File "/usr/local/bin/mvftoms.py", line 72, in load
    da.store([dataset.vis._dask_getitem(indices),
AttributeError: 'DaskLazyIndexer' object has no attribute 'dask_getitem'
Traceback (most recent call last):
  File "/code/run.py", line 32, in <module>
    utils.xrun(cab["binary"], args+files )
  File "/utils/utils/__init__.py", line 74, in xrun
    raise SystemError('%s: returns errr code %d'%(command, process.returncode))
SystemError: mvftoms.py: returns errr code 1
```

I fixed by changing `dask_getitem` to `__getitem__` in lines 72-74 of mvftoms.py

2.  All sources must be dask objects

```
scan   1 (   3 samples) loaded. Target: 'J124906-1941'. Writing to disk...
Traceback (most recent call last):
  File "/usr/local/bin/mvftoms.py", line 791, in <module>
    main()
  File "/usr/local/bin/mvftoms.py", line 541, in main
    scan_vis_data, scan_weight_data, scan_flag_data)
  File "/usr/local/bin/mvftoms.py", line 75, in load
    [vis, weights, flags], lock=False)
  File "/usr/local/lib/python2.7/dist-packages/dask/array/core.py", line 943, in store
    raise ValueError("All sources must be dask array objects")
ValueError: All sources must be dask array objects
Traceback (most recent call last):
  File "/code/run.py", line 32, in <module>
    utils.xrun(cab["binary"], args+files )
  File "/utils/utils/__init__.py", line 74, in xrun
    raise SystemError('%s: returns errr code %d'%(command, process.returncode))
SystemError: mvftoms.py: returns errr code 1
```

:sos:

bmerry, 2018-09-09:
@SpheMakh see if #185 fixes this for you.

closed 2018-09-10 by bmerry

SpheMakh, 2018-09-10:
That did it. Thanks @bmerry

PR #185 (2018-09-10): Fix mvftoms after the oindex update
mvftowas was broken by #180, which took away the `dask_getitem` member
function and replaced it with a `_dask_getitem` free function. Fix by
renaming the free function to `dask_getitem`, importing it into the
top-level katdal namespace, and using it in mvftoms.

Closes #184.
commit 770188e3 (2018-09-10): Fix mvftoms after the oindex update

mvftowas was broken by #180, which took away the `dask_getitem` member
function and replaced it with a `_dask_getitem` free function. Fix by
renaming the free function to `dask_getitem`, importing it into the
top-level katdal namespace, and using it in mvftoms.

Closes #184.