ska-sa/katdal #186: Not converting all scans in archive data to ms and Read timed out

sirothia, 2018-09-11:
mvftoms.py  does not convert all scans in the archive data to ms. Each time I try generates different ms with several scans missing. Also sometimes gives Read timed out error somewhere midway and stop converting. Have tried this on following datasets

1536252666_sdp_l0.rdb 
1536431487_sdp_l0.rdb 
1536438420_sdp_l0.rdb 
1536607816_sdp_l0.rdb

log file for two of conversion using 1536252666_sdp_l0.rdb are attached

[1536252666_std.txt](https://github.com/ska-sa/katdal/files/2370771/1536252666_std.txt)
[1536252666_std_1.txt](https://github.com/ska-sa/katdal/files/2370772/1536252666_std_1.txt)

bmerry, 2018-09-11:
The read timeout is most likely because @ludwigschwardt seems to like setting short timeouts for some reason. Try searching for `timeout=10` in your katdal installation (should be in chunkstore_s3.py) and changing it to something much bigger e.g. `timeout=300`.

I can't imagine what would cause scans to be skipped non-deterministically, but maybe see if increasing the timeout fixes that problem too.

bmerry, 2018-09-17:
@sirothia @spassmoor did the suggested workaround work?

spassmoor, 2018-09-17:
@bmerry  the problem re-occurered for me so I rewrote my data reading function to read a scan (~50 reads) at a time for all antennas instead of originally reading for one Antenna at a time , one scan at a time  ( about ~3000 reads).  This seems to work , also 20 times faster read times.

ludwigschwardt, 2018-09-17:
@sirothia, just be aware that MS conversion only considers scans where the reference antenna **'track'**s an **radec** target, so any 'slew' or 'scan' antenna activity, or azel or satellite targets, are ignored.

ctgschollar, 2018-09-20:
So i tried using mvftoms a quite heavily over the last few days. I found that the timeouts occur whether you use the haproxy or any of the gateways into ceph. However, once I used the suggestion by @bmerry I found I could quite happily convert observations without a problem.

PR #188 (2018-09-20): Increase default timeout for S3ChunkStore
Closes #186.
commit 5bfd4a29 (2018-09-20): Increase default timeout for S3ChunkStore

Closes #186.
closed 2018-09-20 by bmerry