ska-sa/katdal #190: Getting corrupt MS from mvftoms

SpheMakh, 2018-09-25:
I tried to inspect the MS using CASA listobs task and got the following error:

The mvf file I used is http://stgr1.sdp.mkat.chpc.kat.ac.za:7480/1535789408/1535789408_sdp_l0.full.rdb 

```
2018-09-25 13:57:32	INFO	listobs::::	
2018-09-25 13:57:32	INFO	listobs::::+	##########################################
2018-09-25 13:57:32	INFO	listobs::::+	##### Begin Task: listobs            #####
2018-09-25 13:57:32	INFO	listobs::::	listobs(vis="NGC4993_OFF_SEPT0118_RAW.ms",selectdata=True,spw="",field="",antenna="",
2018-09-25 13:57:32	INFO	listobs::::+	        uvrange="",timerange="",correlation="",scan="",intent="",
2018-09-25 13:57:32	INFO	listobs::::+	        feed="",array="",observation="",verbose=True,listfile="",
2018-09-25 13:57:32	INFO	listobs::::+	        listunfl=False,cachesize=50,overwrite=False)
2018-09-25 13:57:33	SEVERE	listobs::ms::close	Exception Reported: Exception: Illegal FIELD_ID value 27 found in main table. /data/sphe/NGC4993_SEPT01/msdir/NGC4993_OFF_SEPT0118_RAW.ms/FIELD only has 0 rows (IDs).
2018-09-25 13:57:33	SEVERE	listobs::ms::close+	... thrown by void casa::MSChecker::checkReferentialIntegrity() const at File: /var/rpmbuild/BUILD/casa-test/casa-test-5.0.101/code/msvis/MSVis/MSChecker.cc, line: 78
2018-09-25 13:57:33	SEVERE	listobs::::	*** Error *** Exception: Illegal FIELD_ID value 27 found in main table. /data/sphe/NGC4993_SEPT01/msdir/NGC4993_OFF_SEPT0118_RAW.ms/FIELD only has 0 rows (IDs).
2018-09-25 13:57:33	SEVERE	listobs::::+	... thrown by void casa::MSChecker::checkReferentialIntegrity() const at File: /var/rpmbuild/BUILD/casa-test/casa-test-5.0.101/code/msvis/MSVis/MSChecker.cc, line: 78
2018-09-25 13:57:33	SEVERE	ms::detached	ms is not attached to a file - cannot perform operation.
2018-09-25 13:57:33	SEVERE	ms::detached+	Call ms.open('filename') to reattach.
2018-09-25 13:57:33	INFO	listobs::::	##### End Task: listobs              #####
2018-09-25 13:57:33	INFO	listobs::::+	##########################################
```

I've been struggling to convert this file to an MS for almost a month now. Is it possible for someone who more familiar with katdal to get me the MS from it?

SpheMakh, 2018-09-25:
The MS is on imgr-com-1 at

imgr-com-1:/data/sphe/NGC4993_SEPT01/msdir/NGC4993_OFF_SEPT0118_RAW.ms

sratcliffe, 2018-09-25:
I will have a look at this later this evening and see if it can be recovered.

sratcliffe, 2018-09-25:
@SpheMakh - looks like I have generated a valid MS (at least listobs() on it works OK). Busy copying it onto imgr-com-1 now - ETA around 2 hours - will let you know when it is done.

SpheMakh, 2018-09-25:
Thanks! @sratcliffe

sratcliffe, 2018-09-25:
Hey @SpheMakh - the data is available on imgr-com-1:

/data/sphe/NGC4993_SEPT01/input_new

Let me know how it goes...

ludwigschwardt, 2019-05-06:
Did you resolve this, @SpheMakh?

SpheMakh, 2019-05-06:
Yes, it was resolved.

closed 2019-05-06 by SpheMakh