ska-sa/katdal #222: Importing memory module fails

PeterKamphuis, 2019-02-12:
katdal seems to be unable to find katsdptelstate.memory

`meerkathi - 2019-02-12 12:29:45,475 INFO - running: mvftoms.py --output-ms /home/jenkins/msdir/1477074305.ms --tar  --channel-range 2525,2776 --model-data  /input/1477074305.h5
meerkathi - 2019-02-12 12:29:45,475 INFO - Traceback (most recent call last):
meerkathi - 2019-02-12 12:29:45,475 INFO -   File "/usr/local/bin/mvftoms.py", line 42, in <module>
meerkathi - 2019-02-12 12:29:45,476 INFO -     import katdal
meerkathi - 2019-02-12 12:29:45,476 INFO -   File "/usr/local/lib/python2.7/dist-packages/katdal/__init__.py", line 229, in <module>
meerkathi - 2019-02-12 12:29:45,476 INFO -     from .datasources import open_data_source
meerkathi - 2019-02-12 12:29:45,476 INFO -   File "/usr/local/lib/python2.7/dist-packages/katdal/datasources.py", line 29, in <module>
meerkathi - 2019-02-12 12:29:45,476 INFO -     import katsdptelstate.memory
meerkathi - 2019-02-12 12:29:45,476 INFO - ImportError: No module named memory
meerkathi - 2019-02-12 12:29:45,476 INFO - Traceback (most recent call last):
meerkathi - 2019-02-12 12:29:45,476 INFO -   File "/scratch/code/run.py", line 41, in <module>
meerkathi - 2019-02-12 12:29:45,476 INFO -     utils.xrun(cab["binary"], args+files )
meerkathi - 2019-02-12 12:29:45,476 INFO -   File "/scratch/stimela/utils/__init__.py", line 73, in xrun
meerkathi - 2019-02-12 12:29:45,476 INFO -     raise SystemError('%s: returns errr code %d'%(command, process.returncode))`

bmerry, 2019-02-12:
You need to update katsdptelstate to the latest version (Github master). @ludwigschwardt is preparing a new PyPI release of katsdptelstate for this at the moment.

ludwigschwardt, 2019-02-12:
Please try the latest katdal 0.12 (released a few minutes ago).

ludwigschwardt, 2019-04-05:
Has this been resolved?

ludwigschwardt, 2021-08-10:
This died of old age.

closed 2021-08-10 by ludwigschwardt