ska-sa/katdal #237: botocore 403 error with katdal installed on com07

bennahugo, 2019-04-01:
Not sure why this is being raised (other datasets of the same period downloads successfully):
Observation: 1527694287 (2018/05/30 - commissioning dataset):
```
Wrote scan data (2349.207479 MB) in 37.662396 s (62.375412 MBps)

scan  43 (  37 samples) loaded. Target: 'PKS 1934-63'. Writing to disk...
Traceback (most recent call last):
  File "/usr/local/bin/mvftoms.py", line 777, in <module>
    main()
  File "/usr/local/bin/mvftoms.py", line 539, in main
    scan_vis_data, scan_weight_data, scan_flag_data)
  File "/usr/local/bin/mvftoms.py", line 69, in load
    [vis, weights, flags], lock=False)
  File "/usr/local/lib/python2.7/dist-packages/dask/array/core.py", line 949, in store
    result.compute(**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/dask/base.py", line 154, in compute
    (result,) = compute(self, traverse=False, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/dask/base.py", line 407, in compute
    results = get(dsk, keys, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/dask/threaded.py", line 75, in get
    pack_exception=pack_exception, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/dask/local.py", line 521, in get_async
    raise_exception(exc, tb)
  File "/usr/local/lib/python2.7/dist-packages/dask/local.py", line 290, in execute_task
    result = _execute_task(task, data)
  File "/usr/local/lib/python2.7/dist-packages/dask/local.py", line 270, in _execute_task
    args2 = [_execute_task(a, cache) for a in args]
  File "/usr/local/lib/python2.7/dist-packages/dask/local.py", line 267, in _execute_task
    return [_execute_task(a, cache) for a in arg]
  File "/usr/local/lib/python2.7/dist-packages/dask/local.py", line 271, in _execute_task
    return func(*args2)
  File "/usr/local/lib/python2.7/dist-packages/katdal/chunkstore.py", line 127, in func_returning_chunk
    value = func(array_name, slices, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/katdal/chunkstore_s3.py", line 167, in has_chunk
    self.client.head_object(Bucket=bucket, Key=key)
  File "/usr/local/lib/python2.7/dist-packages/botocore/client.py", line 314, in _api_call
    return self._make_api_call(operation_name, kwargs)
  File "/usr/local/lib/python2.7/dist-packages/botocore/client.py", line 612, in _make_api_call
    raise error_class(parsed_response, operation_name)
botocore.exceptions.ClientError: An error occurred (403) when calling the HeadObject operation: Forbidden
```

bennahugo, 2019-04-01:
Retrying now with latest katdal from PyPI

bennahugo, 2019-04-01:
Slightly different trace with the latest release from PyPI, but still a 403. I'm guessing the archive for that particular observation has been corrupted somehow?
```
scan  54 (  36 samples) loaded. Target: 'PKS 1934-63'. Writing to disk...
Traceback (most recent call last):
  File "/home/bhugo/mvftoms/bin/mvftoms.py", line 787, in <module>  
    main()
  File "/home/bhugo/mvftoms/bin/mvftoms.py", line 537, in main
    scan_vis_data, scan_weight_data, scan_flag_data)
  File "/home/bhugo/mvftoms/bin/mvftoms.py", line 71, in load
    out=[vis, weights, flags])
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/katdal/lazy_indexer.py", line 594, in get
    da.store(kept, out, lock=False)
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/dask/array/core.py", line 741, in store
    result.compute(**kwargs)
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/dask/base.py", line 156, in compute
    (result,) = compute(self, traverse=False, **kwargs)
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/dask/base.py", line 398, in compute
    results = schedule(dsk, keys, **kwargs)
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/dask/threaded.py", line 76, in get
    pack_exception=pack_exception, **kwargs)
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/dask/local.py", line 462, in get_async
    raise_exception(exc, tb)
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/dask/local.py", line 230, in execute_task
    result = _execute_task(task, data)
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/dask/core.py", line 118, in _execute_task
    args2 = [_execute_task(a, cache) for a in args]
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/dask/core.py", line 119, in _execute_task
    return func(*args2)
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/katdal/chunkstore.py", line 242, in get_chunk_or_zeros
    return self.get_chunk(array_name, slices, dtype)
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/katdal/chunkstore_s3.py", line 332, in get_chunk
    with self._request(chunk_name, 'GET', url, stream=True) as response:
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/katdal/chunkstore_s3.py", line 324, in _request
    _raise_for_status(response)
  File "/home/bhugo/mvftoms/local/lib/python2.7/site-packages/katdal/chunkstore_s3.py", line 141, in _raise_for_status
    raise StoreUnavailable(str(error))
katdal.chunkstore.StoreUnavailable: 403 Client Error: Forbidden for url: http://archive-gw-1.kat.ac.za:7480/1527694287_sdp_l0/weights_channel/01438_00000.npy
```

ctgschollar, 2019-04-01:
This is an issue with Ceph, not katdal. I can fix it for you.

ctgschollar, 2019-04-01:
Try it again

bennahugo, 2019-04-01:
Hi @ctgschollar. Nope still failing at the same point.

bennahugo, 2019-04-01:
Btw if it is only this archive that you can detect an error for from your end I think it is low priority for fixing. Luckily there was an uncorrupted dataset just the day previous.

lmzxqtom, 2019-04-02:
@bennahugo there was no bucket policy set for the s3 bucket _'1527694287_sdp_l0'_. I've tested and access is now granted.
As to why the policy was not originally set, it's probably before we started using policies to manage bucket access.

bennahugo, 2019-04-03:
@lmzxqtom thanks running it now. Will close the ticket if it runs through.

lmzxqtom, 2019-04-03:
@bennahugo great! Keep me posted.

bennahugo, 2019-04-03:
Yup she now works fine. Not sure why the permissions didn't propogate properly in that one instance, but may just be a planetary alignment issue

bennahugo, 2019-04-03:
Thanks @lmzxqtom

closed 2019-04-03 by bennahugo