ska-sa/katdal #246: the augment script doesn't exist

telegraphroad, 2019-06-05:
When trying to read an h5 file with the open function, I get the error "HDF5 file not augmented - please run augment4.py(provided by k7augment package). There are several similar mentions to augment scripts throughout the code with package names like "k7augment", "k7_augment" and "katsdisp", but I can't seem to find these packages and/or scripts anywhere.

ludwigschwardt, 2019-06-05:
Hi Saba,

Please send me the name of the file and I'll have a look.

telegraphroad, 2019-06-05:
Thanks Ludwig!

File: /katdal/h5datav1.py         Lines 255-257
File: /katdal/h5datav2.py         Lines 386-388

And some 8-9 yo commits

ludwigschwardt, 2019-06-05:
Sorry, I meant the name of the h5 file 😃

telegraphroad, 2019-06-05:
I'm a bit confused, you need the name of the h5 file I'm working on?It is a file I got from my supervisor for my thesis. Anyways, the name is "L559133.MS_extract.h5"

ludwigschwardt, 2019-06-05:
I had hoped you got the file from our archive and I could independently have a look at it, but that file looks unfamiliar to me. I'm guessing it's not in the correct format anymore (especially if "extracted from an MS"), hence the request for augmentation. I would need to know more about it to help you.

telegraphroad, 2019-06-05:
Thanks for the update, these were just to get me started. I will ask for updated files, something not "MS_extract"ed.