ska-sa/katdal #268: Selection mechanism does not limit available scan selection after previous selection

bennahugo, 2019-10-02:
If I'm not missing something here the select method modifies the dataset in place. However, a cascading selection can result in a superset of the original selection. The following ought to be equivalent:
```
n [39]: kd.select(targets=["PKS 0408-65"])

In [40]: kd.select(scans=range(0, 100))

In [41]: list(kd.scans())
Out[41]: 
[(0, 'slew', <katpoint.Target 'PKS 0408-65' body=radec at 0x7fe2e6ef9c90>),
 (1, 'track', <katpoint.Target 'PKS 0408-65' body=radec at 0x7fe2e6ef9c90>),
 (2, 'slew', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (3, 'track', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (4, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (5, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (6, 'slew', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (7, 'track', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (8, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (9, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (10, 'slew', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (11, 'track', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (12, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (13, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (14, 'slew', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (15, 'track', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (16, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (17, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (18, 'slew', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (19, 'track', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (20, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (21, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (22, 'slew', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (23, 'track', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (24, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (25, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (26, 'slew', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (27, 'track', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (28, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (29, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (30, 'slew', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (31, 'track', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (32, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (33, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (34, 'slew', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (35, 'track', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (36, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (37, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (38, 'slew', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (39, 'track', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (40, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (41, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (42, 'slew', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (43, 'track', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (44, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (45, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (46, 'slew', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (47, 'track', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (48, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (49, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (50, 'slew', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (51, 'track', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (52, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (53, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (54, 'slew', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (55, 'track', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (56, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (57, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (58, 'slew', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (59, 'track', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (60, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (61, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (62, 'slew', <katpoint.Target 'PKS 1934-63' body=radec at 0x7fe2e6ec8510>),
 (63, 'track', <katpoint.Target 'PKS 1934-63' body=radec at 0x7fe2e6ec8510>),
 (64, 'slew', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (65, 'track', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (66, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (67, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (68, 'slew', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (69, 'track', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (70, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (71, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (72, 'slew', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (73, 'track', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (74, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (75, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (76, 'slew', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (77, 'track', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (78, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (79, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (80, 'slew', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (81, 'track', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (82, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (83, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (84, 'slew', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (85, 'track', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (86, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (87, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (88, 'slew', <katpoint.Target 'PKS 1934-63' body=radec at 0x7fe2e6ec8510>),
 (89, 'track', <katpoint.Target 'PKS 1934-63' body=radec at 0x7fe2e6ec8510>),
 (90, 'slew', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (91, 'track', <katpoint.Target 'A3528N' body=radec at 0x7fe2e6ef9d50>),
 (92, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (93, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (94, 'slew', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (95, 'track', <katpoint.Target 'A3528S' body=radec at 0x7fe2e6ef9dd0>),
 (96, 'slew', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (97, 'track', <katpoint.Target 'J1311-222' body=radec at 0x7fe2e6ef9d90>),
 (98, 'slew', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>),
 (99, 'track', <katpoint.Target 'A3532' body=radec at 0x7fe2e6ef9e50>)]

In [42]: kd.select(reset='T')

In [43]: kd.select(targets=["PKS 0408-65"], scans=range(0, 100))

In [44]: list(kd.scans())
Out[44]: 
[(0, 'slew', <katpoint.Target 'PKS 0408-65' body=radec at 0x7fe2e6ef9c90>),
 (1, 'track', <katpoint.Target 'PKS 0408-65' body=radec at 0x7fe2e6ef9c90>)]

```

ludwigschwardt, 2019-10-02:
You are missing something... Please read the docs.

From the `DataSet.select` [docstring](https://github.com/ska-sa/katdal/blob/f7d7480a1004658b531dc406109cd0003de54463/katdal/dataset.py#L584):

        If there are multiple criteria on the same dimension within a select()
        call, they are ANDed together, while multiple items within the same
        criterion (e.g. `targets=['Hyd A', 'Vir A']`) are ORed together. When a
        second select() call is done, all new selections replace previous
        selections on the same dimension, while existing selections on other
        dimensions are preserved. The `reset` parameter finetunes this behaviour.

So cascading selects are not meant to be the same as multiple criteria within the same select.

bennahugo, 2019-10-03:
Whoops my bad. I assumed the same behavior as cascading SQL selections. Lets close this one

closed 2019-10-03 by bennahugo