ska-sa/katdal #296: mvtoms.py throws connection reset by peer error

SpheMakh, 2020-03-26:
I'm using katdal 0.15 in an ubuntu 18.04 docker container

```
can   4 ( 599 samples) loaded. Target: 'J1939-6342'. Writing to disk...
Added new field 1: 'J1939-6342' 19:39:25.03 -63:42:45.6
Wrote scan data (201912.166424 MiB) in 2285.384612 s (88.349316 MiBps)

scan   5 ( 602 samples) loaded. Target: 'J1939-6342'. Writing to disk...
Traceback (most recent call last):
  File "/usr/local/bin/mvftoms.py", line 816, in <module>
    main()
  File "/usr/local/bin/mvftoms.py", line 566, in main
    scan_vis_data, scan_weight_data, scan_flag_data)
  File "/usr/local/bin/mvftoms.py", line 92, in load
    out=[vis, weights, flags])
  File "/usr/local/lib/python3.6/dist-packages/katdal/lazy_indexer.py", line 594, in get
    da.store(kept, out, lock=False)
  File "/usr/local/lib/python3.6/dist-packages/dask/array/core.py", line 951, in store
    result.compute(**kwargs)
  File "/usr/local/lib/python3.6/dist-packages/dask/base.py", line 166, in compute
    (result,) = compute(self, traverse=False, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/dask/base.py", line 437, in compute
    results = schedule(dsk, keys, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/dask/threaded.py", line 84, in get
    **kwargs
  File "/usr/local/lib/python3.6/dist-packages/dask/local.py", line 486, in get_async
    raise_exception(exc, tb)
  File "/usr/local/lib/python3.6/dist-packages/dask/local.py", line 316, in reraise
    raise exc
  File "/usr/local/lib/python3.6/dist-packages/dask/local.py", line 222, in execute_task
    result = _execute_task(task, data)
  File "/usr/local/lib/python3.6/dist-packages/dask/core.py", line 121, in _execute_task
    return func(*(_execute_task(a, cache) for a in args))
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore.py", line 243, in get_chunk_or_zeros
    return self.get_chunk(array_name, slices, dtype)
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore_s3.py", line 610, in get_chunk
    headers=headers, stream=True)
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore_s3.py", line 587, in complete_request
    result = process(response)
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore_s3.py", line 173, in _read_chunk
    chunk = read_array(data._fp)
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore_s3.py", line 151, in read_array
    bytes_read = fp.readinto(memoryview(data.view(np.uint8)))
  File "/usr/lib/python3.6/http/client.py", line 503, in readinto
    n = self.fp.readinto(b)
  File "/usr/lib/python3.6/socket.py", line 586, in readinto
    return self._sock.recv_into(b)
  File "/usr/lib/python3.6/ssl.py", line 1012, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/lib/python3.6/ssl.py", line 874, in read
    return self._sslobj.read(len, buffer)
  File "/usr/lib/python3.6/ssl.py", line 631, in read
    v = self._sslobj.read(len, buffer)
ConnectionResetError: [Errno 104] Connection reset by peer
```

SpheMakh, 2020-03-26:
This is the ID of the data I'm trying to download 1584577476

SpheMakh, 2020-04-07:
@ludwigschwardt any ideas?

ludwigschwardt, 2020-04-08:
I could not recreate the error. The server typically slammed down the phone on their side with such an error message, indicating temporary overload which should go away if you try again later.

I'm still a bit sad since my latest improvements aim to catch these errors and turn them into flagged missing data without crashing the script. I'm keeping this open to remind me to catch this error too.

SpheMakh, 2020-04-08:
I've tried a couple of times already, and it fails at the same point each time; at around 500Gb. But I'll give it another go

ludwigschwardt, 2020-04-08:
Interesting... I tried the following:
```
import katdal
from katdal.lazy_indexer import DaskLazyIndexer

d = katdal.open('...')
d.select(scans=5)   # which is where you are getting stuck
v, w, f = DaskLazyIndexer.get([d.vis, d.weights, d.flags], 0)
for n in range(602):
    print(n)
    DaskLazyIndexer.get([d.vis, d.weights, d.flags], n, out=[v, w, f])
```
It made it all the way to the end... Maybe try this on your setup.

bennahugo, 2020-04-08:
Update your katdal Sphe. If you are coming in from Rhodes you need the
latest and greatest! Had run into this network endpoint problem before

-- 
--
Benjamin Hugo

PhD. student,
Centre for Radio Astronomy Techniques and Technologies
Department of Physics and Electronics
Rhodes University

Junior software developer
Radio Astronomy Research Group
South African Radio Astronomy Observatory
Black River Business Park
Observatory
Cape Town

ludwigschwardt, 2020-04-08:
katdal 0.15 is pretty new (just pre-lockdown).

SpheMakh, 2020-05-20:
This only happens when I'm running in a docker container. It works fine outside a container. @ludwigschwardt are there any containers that use mvftoms that you know of, maybe I made a mistake in building mine.

spassmoor, 2020-05-20:
I have not had this issue running on my  docker container that has mvftoms.py . I should also point out  that I run it on one of the comm machines and this would mitigate any bad network problems.

SpheMakh, 2020-05-20:
@spassmoor I'm running on com08. Can you share the Dockerfile?

spassmoor, 2020-05-20:
[docker.zip](https://github.com/ska-sa/katdal/files/4656143/docker.zip)

Is this what you wanted ?

SpheMakh, 2020-05-20:
Thanks, I'll give it go.

ludwigschwardt, 2020-05-20:
Just remember that this is a public thread, in case your zip contains sensitive info :-)

spassmoor, 2020-05-20:
Other than my work email address and my preferred version of tornado I don't think there is anything sensitive in it.

sarrvesh, 2021-02-25:
Hey folks, I encountered the same issue using katdal 0.17. mvftoms.py failed for me with a connection time out error on my dataset (1596945366). I tried running @ludwigschwardt 's script above and that failed with the same time out error. Any ideas how to solve this issue?

bennahugo, 2021-02-25:
Need more info here.

Is this inside a container? If so please check that your docker bridge is
working properly by pinging or telnet inside your container.

If it is working can you indicate whether it fails part way through
download or right at the start. There might be a misconfigured end point or
something else not related to containerization.

sarrvesh, 2021-02-25:
I get the same error in a container environment and in a normal virtualenv installation.  It seems to fail right away with the following error:

```
StoreUnavailable: Chunk '1596945366-sdp-l0/correlator_data/00168_00000_00000': HTTPConnectionPool(host='archive-gw-1.kat.ac.za', port=7480): Max retries exceeded with url: /1596945366-sdp-l0/correlator_data/00168_00000_00000.npy (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f7263391860>, 'Connection to archive-gw-1.kat.ac.za timed out. (connect timeout=30)'))
```

My network, otherwise, works just fine.

ludwigschwardt, 2021-02-25:
Hi @sarrvesh, a **connection timeout** indicates that you could not even start to talk to the archive server, i.e. the phone just rings and rings and nobody picks up. This differs from a **connection reset** (the topic of this issue), which is the server slamming down the phone in the middle of your conversation.

I see that you are trying to connect to port 7480. Are you on a machine in the CHPC cluster? If not, you'll need to connect to port 443, aka **https**, and use a token as provided by the RDB link button on the archive. So instead of
```
d = katdal.open('http://archive-gw-1.kat.ac.za:7480/1596945366/1596945366_sdp_l0.full.rdb')
```
try
```
d = katdal.open('https://archive-gw-1.kat.ac.za/1596945366/1596945366_sdp_l0.full.rdb?token=<your-token>')
```
I managed to download dump 168 with both methods just now, so the server is up and your dataset is intact.

sarrvesh, 2021-02-25:
Ah, interesting. Yeah, that works. Thanks very much.

ludwigschwardt, 2021-02-25:
Pleasure!