ska-sa/katdal #309: scans don't apply multiple negated tags, or a mix including at least one negated tag, correctly

adriaanph, 2020-07-07:
currently, select(scans=["~slew","~stop"]) does nothing. that's because the scan filters are ORed together. same with compscans.

the current behaviour is unexpected since it is reasonable to expect the rules of boolean algebra to apply, i.e. all un-negated tags get ORed while negated tags get ANDed with the rest. 

the relevant code appears in katdal/dataset.py line 740, where negated tags are ORed. this OR could simply be changed to AND.

while one may go overboard with this issue and invent a syntax to support complicated sequences of ANDs and ORs, i propose that a reasonable user would be happy to accept that all un-negated tags are ORed together, before being ANDed with the negated tags.

ludwigschwardt, 2020-07-08:
Why not just `select(scans=['track', 'scan'])`? Or is this trickier with compscans?

adriaanph, 2020-07-08:
indeed, there are many work-arounds possible, but the issue remains.