ska-sa/katdal #312: IndexError in mvftoms.py

o-smirnov, 2020-09-01:
Happens both with current master and v0.16:

```
$ mvftoms.py -f --flags=cam 1564201853_sdp_l0.rdb 1564356988_sdp_l0.rdb
...

Traceback (most recent call last):
  File "/home/oms/.venv/katdal/bin/mvftoms.py", line 816, in <module>
    main()
  File "/home/oms/.venv/katdal/bin/mvftoms.py", line 477, in main
    scan_vis_data = np.empty(in_chunk_shape, dataset.vis.dtype)
  File "/home/oms/.venv/katdal/lib/python3.6/site-packages/katdal/concatdata.py", line 634, in vis
    return ConcatenatedLazyIndexer([d.vis for d in self.datasets])
  File "/home/oms/.venv/katdal/lib/python3.6/site-packages/katdal/concatdata.py", line 71, in __init__
    self.indexers = [indexer for indexer in indexers if indexer.shape[0]]
  File "/home/oms/.venv/katdal/lib/python3.6/site-packages/katdal/concatdata.py", line 71, in <listcomp>
    self.indexers = [indexer for indexer in indexers if indexer.shape[0]]
  File "/home/oms/.venv/katdal/lib/python3.6/site-packages/katdal/lazy_indexer.py", line 624, in shape
    return self.dataset.shape
  File "/home/oms/.venv/katdal/lib/python3.6/site-packages/katdal/lazy_indexer.py", line 523, in dataset
    dataset = dask_getitem(self._orig_dataset, self.keep)
  File "/home/oms/.venv/katdal/lib/python3.6/site-packages/katdal/lazy_indexer.py", line 131, in dask_getitem
    indices = _simplify_index(indices, x.shape)
  File "/home/oms/.venv/katdal/lib/python3.6/site-packages/katdal/lazy_indexer.py", line 78, in _simplify_index
    indices = da.slicing.normalize_index(indices, shape)
  File "/home/oms/.venv/katdal/lib/python3.6/site-packages/dask/array/slicing.py", line 827, in normalize_index
    check_index(i, d)
  File "/home/oms/.venv/katdal/lib/python3.6/site-packages/dask/array/slicing.py", line 883, in check_index
    % (x.size, dimension)
IndexError: Boolean array length 7320 doesn't equal dimension 7080
```

o-smirnov, 2020-09-01:
Ahh, perhaps this is user error. I specified two datasets, which it is clearly unable to concatenate. Running one at a time seems to work.

closed 2020-09-01 by o-smirnov

ludwigschwardt, 2020-09-01:
Thanks, I'll still have a look and see if I can improve the error reporting.

o-smirnov, 2020-09-01:
Come to think of it, I'm not sure what concatenation exactly means in this context. If it's the same as concatenating two separate MSs, then I don't see why it should fail like this. But maybe something else is meant?