ska-sa/katdal #313: Pickle valued telescope state

MziyandaAstro, 2020-09-29:


MziyandaAstro, 2020-09-29:
Hi Guys.

I'm working on I'm trying to use katdal but the first run gives off an error 
"The telescope state contains pickled values. This is a security risk, so is disabled by default. If you trust the source of the data, you can allow the pickles to be loaded by setting KATSDPTELSTATE_ALLOW_PICKLE=1 in the environment. This is needed for MeerKAT data up to March 2019."
I'm not too sure what this means . Can someone please help me out on this.

Cheers
Mziyanda

ludwigschwardt, 2020-09-29:
Hi Mziyanda,

If you want to open older data files with a recent katdal, run the following shell command on the command line before running your script / ipython:
```Bash
export KATSDPTELSTATE_ALLOW_PICKLE=1
```
You can also add the line to your `~/.bashrc` or `~/.bash_profile` config file if you want to enable it all the time.

MziyandaAstro, 2020-09-29:
Thanks alot Ludwig.

CheersMzi

closed 2020-09-30 by ludwigschwardt