ska-sa/katdal #334: cityhash issue with windows

adriaanph, 2021-05-17:
it's not possible to install cityhash under windows, unless one makes a few minor code modifications to it.
see
<https://github.com/tensorflow/datasets/issues/690>
<https://github.com/escherba/python-cityhash/pull/25>

unfortunately the cityhash maintainer doesn't seem to want to incorporate the required changes.

could you look into replacing cityhash with an alternative, perhaps like the tensorflow team, siphash / siphashc / csiphash?

bmerry, 2021-05-17:
Does xxhash work on Windows? We added cityhash as a dependency because Dask needs some package for efficiently hashing arrays (otherwise it uses a very slow fallback), but Dask also supports xxhash, and if one uses the latest of everything then xxhash should be faster too (https://github.com/dask/dask/pull/7086).

adriaanph, 2021-05-17:
yes, seems to work fine (windows 10, python 3.9).

bmerry, 2021-05-17:
My suggestion is that for now we should just change the dependency on cityhash to exclude Windows, and you can install xxhash  instead if you find that things are slow.