ska-sa/katdal #346: Populate SIGMA_SPECTRUM

bennahugo, 2021-09-15:
@landmanbester found the mvftoms populated (Wishart) WEIGHT_SPECTRUM is a very good reflection of the correct weights. However, there is an issue if casa (>=4.4) is used to transfer calibrate / self calibrate.
According to https://casa.nrao.edu/casadocs/casa-5-1.2/reference-material/data-weights the weights are going to reinitialized based on SIGMA (or SIGMA_SPECTRUM), so we should not just have that inverse filled correctly here to make calWt work correctly in applycal. Currently, the SIGMA_SPECTRUM is set to unity, so all the hard work @ludwigschwardt has done is being undone under the hood if CASA is used. This does not affect calibration performed by cubical/quartical at present because the cal weights are not being written to MS. This should be a very simple modification to make in mvftoms

@IanHeywood @o-smirnov @JSKenyon FYI

bennahugo, 2021-09-15:
So after https://github.com/ska-sa/katdal/blob/master/katdal/ms_extra.py#L374 we need to add 
```python
main_dict['SIGMA_SPECTRUM'] = 1.0 / weight_data**0.5
```
Along with a data descriptor for SIGMA_SPECTRUM

ludwigschwardt, 2021-09-15:
Questions:
  - What is the relationship between SIGMA and SIGMA_SPECTRUM?
  - Could I just populate SIGMA_SPECTRUM?
  - If both are provided, are they multiplied together or does SIGMA_SPECTRUM supercede SIGMA?
  
Maybe I should write both the average SIGMA / WEIGHT and normalised frequency-dependent SIGMA_SPECTRUM / WEIGHT_SPECTRUM...

bennahugo, 2021-09-15:
@ludwigschwardt I'm submitting a PR soon, you can review. WEIGHT is just a single value for the entire spectrum per correlation per row, so an average is probably as good as anything. They are not multiplied _SPECTRUM supercedes the singular columns.

bmerry, 2021-09-15:
I'm guessing that adding SIGMA_SPECTRUM will make the measurement set significantly larger. What happens if we instead delete the SIGMA column? Is it required? Will CASA behave sensibly without it?

bennahugo, 2021-09-15:
no SIGMA and WEIGHT are compulsory columns according to memo 229

ludwigschwardt, 2021-09-15:
Or delete WEIGHT_SPECTRUM and let it be initialised from SIGMA_SPECTRUM?

bennahugo, 2021-09-15:
hmm - I will need to test whether that will work. I think the safest assumption is to have both - we certainly do not reinitialize WEIGHT_SPECTRUM and most imaging packages do not initialize it either.

ludwigschwardt, 2021-09-15:
I haven't even heard of SIGMA_SPECTRUM before, so I'm not holding my breath for wide support in CASA land 😂

bennahugo, 2021-09-15:
It is in memo 229 as optional column. For a long time we were wondering why our weight spectrum was being screwed up with casa calibration, so yup - bit obscure but yes calibration assumes it is there when WEIGHT_SPECTRUM is present and imaging assumes either WEIGHT or WEIGHT_SPECTRUM is present - one of the latter two adjusted during calibration. IMAGING_WEIGHTS is completely non-standard but typically used to store weights derived by the imager

JSKenyon, 2021-09-15:
SIGMA_SPECTRUM seems to be optional. To the best of my knowledge, SIGMA is being removed in MSv3 and WEIGHT is being redefined as WEIGHT_SPECTRUM. SIGMA and WEIGHT are completely redundant, provided everything is handled consistently. So the problem in CASA is likely just an implementation detail.

bennahugo, 2021-09-15:
Unfortunately it seems if WEIGHT_SPECTRUM is missing and only SIGMA_SPECTRUM is available then CASA (5.1) do not detect that it can derive WEIGHT_SPECTRUM during applycal. So if you use them they should **BOTH** be there

closed 2021-09-15 by ludwigschwardt

bennahugo, 2022-04-14:
> @bennahugo <https://github.com/bennahugo> have you tested later versions
> of CASA? Isn't 5.1 quite old now? It would seem strange if WEIGHT_SPECTRUM
> is derived from SIGMA_SPECTRUM, but it expects the latter to exist first.
> I'm chatting with @ludwigschwardt <https://github.com/ludwigschwardt>
> about potentially dropping WEIGHT_SPECTRUM to save on disk space (and
> similar for IMAGING_WEIGHT)

No, I have not tested it with casa V6. Do you mind testing that? You can
simply run removecols from casacore.tables on the column and see if
applycal reconstructs it.

My concern with removing it is that older CASA versions do not correctly
handle the spectral weight cases if dropped as discussed on the original
ticket.

I have no problem with dropping imaging_weight here -- it is for the most
part a user-defined column (e.g. to store calculated robust weights).

landmanbester, 2022-04-14:
I think the point of having both WEIGHT_SPECTRUM and SIGMA_SPECTRUM is to account for the change in noise properties when making CORRECTED_DATA (as you should when applying gain solutions with non-unity amplitudes). SIGMA_SPECTRUM should refer to the standard deviation of the thermal noise contribution whereas WEIGHT_SPECTRUM contains the weights pertaining to CORRECTED_DATA. These can always be recomputed given SIGMA_SPECTRUM and the gain solutions but most imagers don't support that

ludwigschwardt, 2022-04-14:
The question is whether WEIGHT_SPECTRUM should be there from the start (taking up unnecessary space), or whether CASA can fill it in later via e.g. applycal.

commit 1b1bf54c (2021-09-15): Populate SIGMA_SPECTRUM and redo SIGMA and WEIGHT (#347)

Add the missing SIGMA_SPECTRUM column (expected by CASA
calibration routines in tandem with WEIGHT_SPECTRUM).

Redo SIGMA and WEIGHT to be average values over the spectrum
instead of just ones.

Fixes #346.