ska-sa/katdal #362: mvftoms produced malformed MS (TiledStMan::headerFileGet: mismatch in #row)

IanHeywood, 2023-01-30:
Trying to use `mvftoms.py` to pull 32k data from the archive to IDIA. The connection frequently drops with:

```
socket.timeout: The read operation timed out
```

but when I restart it continues. It's got most of the way there after a couple of days (11 TB downloaded), but now I'm getting the error below. I get the same error when I try to access the MS using other software. Any tips on how to recover this gratefully received. I'll start the transfer again in the meantime.

Cheers.

```
/users/ianh/venv/katdal/lib/python3.8/site-packages/katdal/applycal.py:156: RuntimeWarning: invalid value encountered in reciprocal
  corrections.append(ComparableArrayWrapper(np.reciprocal(bp)))
/users/ianh/venv/katdal/lib/python3.8/site-packages/katdal/applycal.py:204: RuntimeWarning: invalid value encountered in reciprocal
  return np.reciprocal(smooth_gains)
The following calibration products will be applied: l1.K, l1.B, l1.G, l2.GPHASE
Per user request the following antennas will be selected: 'm000', 'm001', 'm002', 'm003', 'm004', 'm005', 'm006', 'm007', 'm008', 'm010', 'm012', 'm013', 'm014', 'm015', 'm016', 'm017', 'm018', 'm019', 'm020', 'm021', 'm022', 'm023', 'm024', 'm025', 'm026', 'm027', 'm028', 'm029', 'm030', 'm031', 'm032', 'm033', 'm034', 'm035', 'm036', 'm037', 'm038', 'm039', 'm041', 'm042', 'm043', 'm044', 'm045', 'm046', 'm047', 'm048', 'm049', 'm050', 'm051', 'm052', 'm053', 'm054', 'm055', 'm056', 'm057', 'm058', 'm059', 'm060', 'm061', 'm062', 'm063'
Per user request the following target fields will be selected: 'COSMOS_3'
Per user request the following scans will be dumped: 34, 8, 42, 14, 48, 22, 28
Extract MS for spw 0: centre frequency 1284000000 Hz
Will create MS output in 1622376680_sdp_l2.full.ms

#### Producing a full polarisation MS (HH,HV,VH,VV) ####

Using array as the reference antenna. All targets and scans will be based on this antenna.

Iterating through scans in dataset(s)...

Traceback (most recent call last):
  File "/users/ianh/venv/katdal/bin/mvftoms.py", line 916, in <module>
    main()
  File "/users/ianh/venv/katdal/bin/mvftoms.py", line 545, in main
    with ms_extra.open_table(ms_name, verbose=options.verbose) as t:
  File "/users/ianh/venv/katdal/lib/python3.8/site-packages/katdal/ms_extra.py", line 42, in open_table
    return tables.table(name, readonly=readonly, ack=verbose, **kwargs)
  File "/users/ianh/venv/katdal/lib/python3.8/site-packages/casacore/tables/table.py", line 373, in __init__
    Table.__init__(self, tabname, lockopt, opt)
RuntimeError: Table DataManager error: Internal error: TiledStMan::headerFileGet: mismatch in #row; expected 0, found 5313710
```

ludwigschwardt, 2023-01-30:
Sorry for your struggles! 

>  It's got most of the way there after a couple of days (11 TB downloaded),

Do you mean you have a partial MS after a couple of restarts, but now the MS is stuck after the latest restart?

The header probably has 5313710 rows on disk, but I'm still trying to understand why the DataManager expects an empty table...

IanHeywood, 2023-01-31:
> Sorry for your struggles!

No apologies necessary. I'm shamefully late when it comes to using this method, and it's already changing the game. I was looking for the 'help wanted' tag for this issue because I'm not sure it's actually a bug.

> Do you mean you have a partial MS after a couple of restarts, but now the MS is stuck after the latest restart?

The MS is 11 out of maybe 13 TB at this point, so it's annoyingly close. The socket.timeout thing is endemic though, there have been a lot of restarts for this particular observation and some others that seem to be OK. This method seems way faster than what I've been doing before, but I have to babysit it constantly because of the timeouts. I was looking to implement a (perhaps slightly dodgy) script that captures the mvftoms.py output and checks the last line for the time out error and then restarts the transfer if it's there. But this is a new one, it looks like the MS is fundamentally mangled...

> The header probably has 5313710 rows on disk, but I'm still trying to understand why the DataManager expects an empty table...

...and it it looks like some deep casacore lore that I don't know how to diagnose.

Cheers.

bennahugo, 2023-01-31:
@IanHeywood - I know this is not a fix but if you look at the help options
you will see that you can select by scans and fields (rather than
downloading 11TiB in one go?)

Ruby used this previously to get 32k data to James so this may be a
solution for you. What I typically do when downloading to Rhodes is
splitting by field (CASA will allow you to transfer calibrate across
databases) - or you can virtual concat

Let me know,

-- 
--
Benjamin Hugo

PhD. student,
Centre for Radio Astronomy Techniques and Technologies
Department of Physics and Electronics
Rhodes University

Junior software developer
Radio Astronomy Research Group
South African Radio Astronomy Observatory
Black River Business Park
Observatory
Cape Town

ludwigschwardt, 2023-01-31:
When an exception occurs the MS table should have been closed properly but maybe something went wrong there.

I see in the casacore code that there is a TABLEREPAIR option that would allow you to override the row number, but you have to recompile casacore for that (and not sure how well it works at all).

I have an open ticket to address `socket.timeout` issues, but wasn't aware of how often they occur. I'll bump up the priority!

bennahugo, 2023-01-31:
My thoughts on the shape issue is that we are using (or were using) fixed
sized columns. If the correct sizes are not set in the descriptors at the
start it may be part of the problem.

One could switch to variable sized columns but we found early on (back in
the 32-single correlator days) that not using fixed shaped descriptors
leads to an almost an order of magnitude slowdown in the casacore table
system which is not really feasible for reductions

-- 
--
Benjamin Hugo

PhD. student,
Centre for Radio Astronomy Techniques and Technologies
Department of Physics and Electronics
Rhodes University

Junior software developer
Radio Astronomy Research Group
South African Radio Astronomy Observatory
Black River Business Park
Observatory
Cape Town

ludwigschwardt, 2023-01-31:
Thanks @bennahugo. The `mvftoms` script typically opens the table and then adds some rows, one dump at a time. It can now resume, which means that it checks the number of rows in the existing MS and continues from there. Somewhere in the process the row counter was trashed, if you ask me.

IanHeywood, 2023-01-31:
> I know this is not a fix but if you look at the help options you will see that you can select by scans and fields (rather than downloading 11TiB in one go?)

Thanks @bennahugo, @ludwigschwardt. 

I'm already selecting only the target field since it's transferring L2 data. Splitting by scan might be an option, although I'll have to modify the processing scripts and up my bookkeeping game. Having the twin curse of being both busy and lazy I'd rather avoid it.

On that note I've just removed the broken MS and restarted the transfer. It's already almost half way.

Cheers.

IanHeywood, 2023-01-31:
> I have an open ticket to address socket.timeout issues, but wasn't aware of how often they occur. I'll bump up the priority!

I did search for this prior to starting a new issue, but couldn't find it. Can you please link it here if you get a moment?

Thanks!

ludwigschwardt, 2023-01-31:
It's an internal ticket: https://skaafrica.atlassian.net/browse/SPR1-813 (but maybe you could still access it, since I see you have a Jira account).

IanHeywood, 2023-01-31:
I don't have permission, but don't worry about it. Was just going to follow the developments. Cheers.

Jordatious, 2023-02-01:
@ludwigschwardt, this is the same issue we discussed in November. My error was much the same:

> RuntimeError: Table DataManager error: Internal error: TiledStMan::headerFileGet: mismatch in #row; expected 735660, found 739320

For me, the issue arose from the SLURM job timing out, so the script(s) didn't have a chance to close the table gracefully. From memory, your suggested solution was to update the metadata so that it either expects the higher number of rows (assuming they're intact), or remove the newer rows beyond what's expected, so that they can be written again. How possible are either of those solutions?

> I was looking to implement a (perhaps slightly dodgy) script that captures the mvftoms.py output and checks the last line for the time out error and then restarts the transfer if it's there.

I've given @IanHeywood a copy of my method for doing this with SLURM, which uses the requeue option. If it's generally useful for other users, I can post it somewhere. What do you think?

IanHeywood, 2023-02-17:
Just a note that the timeout issue seems to be an actual showstopper right now. Whereas previously it might do a few scans, timeout and then resume, for the last 1-2 days it's been timing out before it manages to complete the transfer of a single scan, thus restarting does not advance things.

Jordatious, 2023-02-17:
I'm also having trouble, although I seem to have gotten further (dump 520, 77, 225). I'm on my 3rd attempt to transfer the same MS over the past few days, which initially wrote 1.1 TB before continually failing, 188 GB, and just now, 551 GB.

ludwigschwardt, 2023-02-17:
I'm very sorry to hear that! I'll see if I can cook up a solution in the next few days.

IanHeywood, 2023-02-20:
Getting somewhat off-topic now, but just wanted a record of this new flavour of download interruption:

```
  File "/users/ianh/venv/katdal/lib/python3.8/site-packages/katsdptelstate/telescope_state.py", line 172, in load_from_file
    keys_loaded = self._backend.load_from_file(file)
  File "/users/ianh/venv/katdal/lib/python3.8/site-packages/katsdptelstate/memory.py", line 177, in load_from_file
    return rdb_reader.load_from_file(MemoryCallback(self._data), file)
  File "/users/ianh/venv/katdal/lib/python3.8/site-packages/katsdptelstate/rdb_reader.py", line 84, in load_from_file
    _parse_rdb_file(parser, callback, cast(BinaryIO, file))
  File "/users/ianh/venv/katdal/lib/python3.8/site-packages/katsdptelstate/rdb_reader.py", line 50, in _parse_rdb_file
    raise RdbParseError(filename) from exc
katsdptelstate.errors.RdbParseError: Invalid RDB file object
```

It looks like the RDB token has spontaneously changed. Don't know if that's expected behaviour!

Cheers.

ludwigschwardt, 2023-02-20:
I think what's happening is that the RDB file itself is truncated during download and therefore could not open. The token should still be valid.

I currently do not retry the RDB download... It's also on the to-do list as a separate but related issue.

ludwigschwardt, 2023-02-20:
Reopening this while it is being fixed.

Jordatious, 2023-04-05:
Hi @ludwigschwardt, I had a go of a remote `mvftoms.py` pull from ilifu using [this branch](https://github.com/ska-sa/katdal/tree/spr1-813-handle-read-errors) of katdal last month after I noticed your pull request recently, and found it worked quite well, even for large pulls (5.7 TiB and 2.4 TiB, requeued 89 and 56 times, respectively). 

I've just started to use it again now, and immediately had the following error thrown upon trying to `import numba`. I'm wondering if something has changed that might cause this error, either on your side or ours (e.g. we recently migrated a number of our VMs to bare metal machines). I started using a more recent stable version of katdal (0.20.2.dev1621+master.eb8db22, using numba-0.56.4 and numpy-1.23.4) and it started pulling immediately. I've also tried updating numba for my venv of katdal using the dev branch, which was unchanged as version 0.56.4, but rolled back to numpy-1.23.5 (from 1.24.2) and installed style-1.1.0 and update-0.0.1, and that also immediately started pulling. If you have any clarity on the issue, please let me know!

```
Traceback (most recent call last):
  File "/idia/users/jcollier/software/katdal-dev/bin/mvftoms.py", line 35, in <module>
    import numba
  File "/idia/users/jcollier/software/katdal-dev/lib/python3.8/site-packages/numba/__init__.py", line 42, in <module>
    from numba.np.ufunc import (vectorize, guvectorize, threading_layer,
  File "/idia/users/jcollier/software/katdal-dev/lib/python3.8/site-packages/numba/np/ufunc/__init__.py", line 3, in <module>
    from numba.np.ufunc.decorators import Vectorize, GUVectorize, vectorize, guvectorize
  File "/idia/users/jcollier/software/katdal-dev/lib/python3.8/site-packages/numba/np/ufunc/decorators.py", line 3, in <module>
    from numba.np.ufunc import _internal
SystemError: initialization of _internal failed without raising an exception
```

bmerry, 2023-04-06:
Sounds like https://github.com/numba/numba/issues/8615

Jordatious, 2023-04-06:
Ok thanks @bmerry. I ended up using the stable version I have of katdal and two ~1.6 TiB MS transfers completed after only ~4 times of having to requeue!

Jordatious, 2023-05-04:
Hi all, just a quick update on this. This week I pulled 3 MSs to ilifu using `mvftoms.py` from the #363 [dev branch](https://github.com/ska-sa/katdal/tree/spr1-813-handle-read-errors) (with numpy-1.23.5), and all 3 completed in one attempt! They respectively took roughly 10.5, 11.5 and 20 hours to pull roughly 1, 1.1 and 2.6 TiB.

ludwigschwardt, 2023-05-04:
Excellent news! I think it's time to merge the PR.

ludwigschwardt, 2024-03-18:
This was fixed by #363.

closed 2024-03-18 by ludwigschwardt