ska-sa/katdal #364: mvtoms.py: AttributeError: module 'numpy' has no attribute 'bool'.

mpatir, 2023-05-11:
Hi developers.

I am trying to retrieve the 32k data from the SARAO archive to IDIA. 

I installed katdal on IDIA as follows:

` srun --X11 --cpus-per-task=2 --mem=16GB --pty bash
 virtualenv -p python3 /scratch3/users/mpati/virtualenvk/katdal
 source /scratch3/users/blah/virtualenvk/katdal/bin/activate
 pip install katdal
 pip install python-casacore
 deactivate
` 

I am running katdal as below:

 ` source /scratch3/users/blah/virtualenvk/katdal/bin/activate
 export KATSDPTELSTATE_ALLOW_PICKLE=1
 mvtoms.py <token-link> -v -C 19829,21551 -p HH,VV --flags '' -o outputfile.ms
` 

 I get the error below: The pip installed katdal seems to be incompatible with something. In any case, how can we fix this issue?
 
`  /scratch3/users/blah/virtualenvk/katdal/lib/python3.8/site-packages/katdal/ms_extra.py:97: FutureWarning: In the future `np.bool` will be defined as the corresponding NumPy scalar.
  'BOOLEAN': np.bool,
Traceback (most recent call last):
  File "/scratch3/users/blah/virtualenvk/katdal/bin/mvftoms.py", line 39, in <module>
    from katdal import averager, ms_async, ms_extra
  File "/scratch3/users/blah/virtualenvk/katdal/lib/python3.8/site-packages/katdal/ms_async.py", line 36, in <module>
    from . import ms_extra
  File "/scratch3/users/blah/virtualenvk/katdal/lib/python3.8/site-packages/katdal/ms_extra.py", line 97, in <module>
    'BOOLEAN': np.bool,
  File "/scratch3/users/blah/virtualenvk/katdal/lib/python3.8/site-packages/numpy/__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'bool'.
`np.bool` was a deprecated alias for the builtin `bool`. To avoid this error in existing code, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
`

mpatir, 2023-05-12:
Hi Ludwig, 
Thank you so much for having a look at this issue. 
Okay, I have just installed master for now (pip install git+<master>) while we wait for the new release.
It works great now. 
Cheers

closed 2023-05-12 by mpatir

ludwigschwardt, 2023-05-18:
The fix has been released in version 0.21.