ska-sa/katdal #367: OutOfRangeError when changing the projection in datasets

spassmoor, 2023-08-31:
When changing the projection on datasets with slews in them, the error  **OutOfRangeError: Target point more than 0.5 pi radians away from reference point**  is produces when there is a slew of that size. 

 I think this is a bug in katpoint that it can come to a sensible answer but a workaround would be set the `katpoint.projection.set_out_of_range_treatment` to either clip or nan rather than raise in katdal.

ludwigschwardt, 2023-09-01:
Thanks Sean! I don't think it's a bug - the "user" (in this case katdal) has to decide what to do when out of range. The default is to be very strict and crash. Setting the out-of-range treatment is not a workaround but exactly this decision to be made. I'll transfer this to an SPR1 ticket.

ludwigschwardt, 2023-10-12:
The relevant ticket is SPR1-2958.

ludwigschwardt, 2026-10-22:
Finally fixed this in #399 😅 The default is now "nan" but it can be set per dataset.

closed 2026-10-22 by ludwigschwardt