ska-sa/katdal #368: Replace underscores with dashes when loading buckets from RDBs

ctgschollar, 2023-09-28:
The new Ceph installation at MeerKAT has done away with underscores in bucket names. This is to make the ceph s3 comply with the s3 protocol.

That means that some of our legacy data from before 2019 has had their buckets renamed, but the RDBs have not been updated to reflect that.

Eg. http://archive-gw-1.kat.ac.za:7480/1540580961_sdp_l1_flags is now http://archive-gw-1.kat.ac.za:7480/1540580961-sdp-l1-flags

Link to the example in the web-archive. https://archive.sarao.ac.za/search/1540580961/ This data is currently in ceph, so a good test.

We need to make a fix for katdal to replace underscores with dashes for all buckets accessed directly from the archive.

I am happy to test this out on our transfer nodes once you have an update for this.

ludwigschwardt, 2023-09-28:
Thanks Chris! Could you copy and paste this into an SPR1 JIRA ticket as well?

ctgschollar, 2023-10-02:
Here you go
https://skaafrica.atlassian.net/browse/SPR1-2990

spassmoor, 2023-10-11:
Hi @ludwigschwardt  This is quite urgent now as the Galactic plane legacy survey us about to be released/published and a large chunk of that data is pre-2019

ludwigschwardt, 2024-05-07:
Fixed by #370.

closed 2024-05-07 by ludwigschwardt