ska-sa/katdal #375: ms does not exist and cannot be created

TariqBlecher, 2024-02-02:
Hi, when trying to download an MS from the MeerKAT archive using the mvftoms.py script, I get the error:

` RuntimeError: Directory: /net/com08/data7/ms_test.ms does not exist and cannot be created
` 
 
 Here's the full output
```

(.venv) tblecher@com07:/net/com08/data7$ /home/tblecher/katdal/scripts/mvftoms.py -o ms_test.ms --flags static,cam,data_lost,ingest_rfi -C 163,3885 --dumptime 8 --chanbin 4 --quack 1 --applycal all https://archive-gw-1.kat.ac.za/1706776111/1706776111_sdp_l0.full.rdb?token=eyJ0eXAiOiJKV1QiLCJhbGciOiJFUzI1NiJ9.eyJpc3MiOiJrYXQtYXJjaGl2ZS5rYXQuYWMuemEiLCJhdWQiOiJhcmNoaXZlLWd3LTEua2F0LmFjLnphIiwiaWF0IjoxNzA2ODY1NDE4LCJwcmVmaXgiOlsiMTcwNjc3NjExMSJdLCJleHAiOjE3MDc0NzAyMTgsInN1YiI6InRibGVjaGVyQHNhcmFvLmFjLnphIiwic2NvcGVzIjpbInJlYWQiXX0.FNN54abuPmwl_Xb2Q4ul6GFn04gBbNE50flPBt3GbDQIKNegD3IftQ0y67_rheQt367716c6DFVAzbFF_VH4QA
/home/tblecher/mlcode/.venv/lib/python3.9/site-packages/katdal/applycal.py:156: RuntimeWarning: invalid value encountered in reciprocal
  corrections.append(ComparableArrayWrapper(np.reciprocal(bp)))
The following calibration products will be applied: l1.K, l1.B, l1.G
Per user request the following antennas will be selected: 'm000', 'm001', 'm002', 'm003', 'm004', 'm005', 'm009', 'm010', 'm011', 'm012', 'm013', 'm014', 'm015', 'm016', 'm017', 'm018', 'm019', 'm020', 'm021', 'm022', 'm023', 'm024', 'm025', 'm026', 'm027', 'm028', 'm029', 'm030', 'm031', 'm032', 'm033', 'm034', 'm035', 'm036', 'm037', 'm039', 'm040', 'm041', 'm042', 'm043', 'm044', 'm045', 'm046', 'm048', 'm049', 'm051', 'm052', 'm053', 'm054', 'm055', 'm056', 'm057', 'm058', 'm059', 'm060', 'm061', 'm062', 'm063'
Per user request the following target fields will be selected: 'J1939-6342'
Per user request the following scans will be dumped: 1, 4
Extract MS for spw 0: centre frequency 1284000000 Hz
Will create MS output in ms_test.ms

#### Producing MS with HH,VV polarisation(s) ####

Channel range 163 through 3885.
Averaging 4 channels, output ms will have 930 channels.
The last 3 channels in the data will be dropped during averaging (4 does not divide 3723).
Averaging 7.996616971962617 second dumps to 7.996616971962617 seconds.

Using array as the reference antenna. All targets and scans will be based on this antenna.

Iterating through scans in dataset(s)...

Traceback (most recent call last):
  File "/home/tblecher/katdal/scripts/mvftoms.py", line 916, in <module>
    main()
  File "/home/tblecher/katdal/scripts/mvftoms.py", line 557, in main
    ms_extra.create_ms(ms_name, table_desc, dminfo)
  File "/home/tblecher/mlcode/.venv/lib/python3.9/site-packages/katdal/ms_extra.py", line 48, in create_ms
    with tables.default_ms(filename, table_desc, dm_info) as main_table:
  File "/home/tblecher/mlcode/.venv/lib/python3.9/site-packages/casacore/tables/table.py", line 70, in default_ms
    return table(_default_ms(name, tabdesc, dminfo), _oper=3)
RuntimeError: Directory: /net/com08/data7/ms_test.ms does not exist and cannot be created
```

ludwigschwardt, 2024-02-02:
I think it's a permission issue. Don't you mean `/net/com08/data7/tblecher/ms_test.ms`?

TariqBlecher, 2024-02-05:
ah yes apologies Ludwig -  that was very much a Friday afternoon sort of bug, it's working now.

closed 2024-02-05 by TariqBlecher