ska-sa/katdal #376: mvftoms.py reading time-out on Jupyter core

f-carotenuto, 2024-03-16:
Hello, 
I am trying to run mvftoms.py on a Jupyter node in IDIA to create an .ms file with only three targets (the full observation had many more targets). However, I always end up with a reading time-out error. Any help would be really appreciated!

This is my command on a Half-max node:
`!mvftoms.py --target J1939-6342,J1424-4913,SwiftJ1518  -o import_test.ms --flags cam,data_lost -v https://archive-gw-1.kat.ac.za/1709952327/1709952327_sdp_l0.full.rdb?token=eyJ0eXAiOiJKV1QiLCJhbGciOiJFUzI1NiJ9.eyJpc3MiOiJrYXQtYXJjaGl2ZS5rYXQuYWMuemEiLCJhdWQiOiJhcmNoaXZlLWd3LTEua2F0LmFjLnphIiwiaWF0IjoxNzEwMTgxMzM3LCJwcmVmaXgiOlsiMTcwOTk1MjMyNyJdLCJleHAiOjE3MTA3ODYxMzcsInN1YiI6ImZyYW5jZXNjby5jYXJvdGVudXRvQHBoeXNpY3Mub3guYWMudWsiLCJzY29wZXMiOlsicmVhZCJdfQ.avuv4LoAL8MQKl4FfXj3avDBTV-8WdzpFfvbA8Ds8fFVAjEvYk12S7G-9QIw5WDupDANTyrvM9kWtWZ4DidnIA`

And this is the output
```
No calibration products will be applied
Per user request the following antennas will be selected: 'm000', 'm001', 'm002', 'm003', 'm004', 'm005', 'm006', 'm007', 'm008', 'm009', 'm010', 'm011', 'm012', 'm013', 'm014', 'm015', 'm016', 'm017', 'm018', 'm019', 'm020', 'm021', 'm022', 'm023', 'm024', 'm025', 'm026', 'm027', 'm028', 'm029', 'm030', 'm031', 'm032', 'm033', 'm034', 'm035', 'm036', 'm037', 'm038', 'm039', 'm040', 'm041', 'm042', 'm043', 'm044', 'm045', 'm046', 'm047', 'm048', 'm049', 'm050', 'm051', 'm052', 'm053', 'm054', 'm055', 'm056', 'm057', 'm058', 'm059', 'm060', 'm061', 'm062', 'm063'
Per user request the following target fields will be selected: 'J1939-6342', 'J1424-4913', 'SwiftJ1518'
Per user request the following scans will be dumped: 1, 33, 5, 7, 9, 13
Extract MS for spw 0: centre frequency 1284000000 Hz
Will create MS output in import_test.ms

#### Producing MS with HH,VV polarisation(s) ####

Using array as the reference antenna. All targets and scans will be based on this antenna.

Iterating through scans in dataset(s)...

Writing static meta data...
Table ANTENNA:
  opened successfully
  added 64 rows
  wrote column 'DISH_DIAMETER' with shape (64,)
  wrote column 'FLAG_ROW' with shape (64,)
  wrote column 'MOUNT' with shape (64,)
  wrote column 'NAME' with shape (64,)
  wrote column 'OFFSET' with shape (64, 3)
  wrote column 'POSITION' with shape (64, 3)
  wrote column 'STATION' with shape (64,)
  wrote column 'TYPE' with shape (64,)
  closed successfully
Table FEED:
  opened successfully
  added 64 rows
  wrote column 'ANTENNA_ID' with shape (64,)
  wrote column 'BEAM_ID' with shape (64,)
  wrote column 'BEAM_OFFSET' with shape (64, 2, 2)
  wrote column 'FEED_ID' with shape (64,)
  wrote column 'INTERVAL' with shape (64,)
  wrote column 'NUM_RECEPTORS' with shape (64,)
  wrote column 'POLARIZATION_TYPE' with shape (64, 2)
  wrote column 'POL_RESPONSE' with shape (64, 2, 2)
  wrote column 'POSITION' with shape (64, 3)
  wrote column 'RECEPTOR_ANGLE' with shape (64, 2)
  wrote column 'SPECTRAL_WINDOW_ID' with shape (64,)
  wrote column 'TIME' with shape (64,)
  closed successfully
Table DATA_DESCRIPTION:
  opened successfully
  added 1 rows
  wrote column 'FLAG_ROW' with shape (1,)
  wrote column 'POLARIZATION_ID' with shape (1,)
  wrote column 'SPECTRAL_WINDOW_ID' with shape (1,)
  closed successfully
Table POLARIZATION:
  opened successfully
  added 1 rows
  wrote column 'CORR_PRODUCT' with shape (1, 2, 2)
  wrote column 'CORR_TYPE' with shape (1, 2)
  wrote column 'FLAG_ROW' with shape (1,)
  wrote column 'NUM_CORR' with shape (1,)
  closed successfully
Table OBSERVATION:
  opened successfully
  added 1 rows
  wrote column 'FLAG_ROW' with shape (1,)
  wrote column 'LOG' with shape (1, 1)
  wrote column 'OBSERVER' with shape (1,)
  wrote column 'PROJECT' with shape (1,)
  wrote column 'RELEASE_DATE' with shape (1,)
  wrote column 'SCHEDULE' with shape (1, 1)
  wrote column 'SCHEDULE_TYPE' with shape (1,)
  wrote column 'TELESCOPE_NAME' with shape (1,)
  wrote column 'TIME_RANGE' with shape (1, 2)
  closed successfully
Successful read/write open of default-locked table import_test.ms: 25 columns, 0 rows
scan   1 (  38 samples) loaded. Target: 'J1939-6342'. Writing to disk...
Added new field 0: 'J1939-6342' 19:39:25.03 -63:42:45.6
Traceback (most recent call last):
  File "/usr/local/bin/mvftoms.py", line 916, in <module>
    main()
  File "/usr/local/bin/mvftoms.py", line 666, in main
    scan_vis_data, scan_weight_data, scan_flag_data)
  File "/usr/local/bin/mvftoms.py", line 116, in load
    out=[vis, weights, flags])
  File "/usr/local/lib/python3.6/dist-packages/katdal/lazy_indexer.py", line 591, in get
    da.store(kept, out, lock=False)
  File "/usr/local/lib/python3.6/dist-packages/dask/array/core.py", line 1041, in store
    result.compute(**kwargs)
  File "/usr/local/lib/python3.6/dist-packages/dask/base.py", line 283, in compute
    (result,) = compute(self, traverse=False, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/dask/base.py", line 565, in compute
    results = schedule(dsk, keys, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/dask/threaded.py", line 84, in get
    **kwargs
  File "/usr/local/lib/python3.6/dist-packages/dask/local.py", line 487, in get_async
    raise_exception(exc, tb)
  File "/usr/local/lib/python3.6/dist-packages/dask/local.py", line 317, in reraise
    raise exc
  File "/usr/local/lib/python3.6/dist-packages/dask/local.py", line 222, in execute_task
    result = _execute_task(task, data)
  File "/usr/local/lib/python3.6/dist-packages/dask/core.py", line 121, in _execute_task
    return func(*(_execute_task(a, cache) for a in args))
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore.py", line 145, in __getitem__
    return self.getter(self.array_name, slices, self.dtype, **self.kwargs)
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore.py", line 325, in get_chunk_or_placeholder
    return self.get_chunk(array_name, slices, dtype)
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore_s3.py", line 638, in get_chunk
    headers=headers, stream=True)
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore_s3.py", line 595, in complete_request
    result = process(response)
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore_s3.py", line 162, in _read_chunk
    chunk = read_array(data._fp)
  File "/usr/local/lib/python3.6/dist-packages/katdal/chunkstore_s3.py", line 141, in read_array
    bytes_read = fp.readinto(memoryview(data.view(np.uint8)))
  File "/usr/lib/python3.6/http/client.py", line 507, in readinto
    n = self.fp.readinto(b)
  File "/usr/lib/python3.6/socket.py", line 586, in readinto
    return self._sock.recv_into(b)
  File "/usr/lib/python3.6/ssl.py", line 1012, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/lib/python3.6/ssl.py", line 874, in read
    return self._sslobj.read(len, buffer)
  File "/usr/lib/python3.6/ssl.py", line 631, in read
    v = self._sslobj.read(len, buffer)
socket.timeout: The read operation timed out
```

Jordatious, 2024-03-18:
Hi @f-carotenuto, you may want to make use of the workflow I wrote to resume an `mvftoms.py` pull via SLURM's requeue feature. I've just copied the workflow from a private repository to my own one, which you can find here:

https://github.com/Jordatious/MVF2MS

It's worth keeping in mind that if your project is a supported project on ilifu, you can push the data directly to IDIA via the SARAO archive, which will be much faster and easier in general.

Jordatious, 2024-03-18:
You'll want to replace [this line](https://github.com/Jordatious/MVF2MS/blob/2d049a224114d21be21371439cecd45fa45e1bc9/requeue-MS-transfer.sbatch#L23) with whatever software (e.g. virtual environment) you install (e.g. following [this](https://docs.google.com/document/d/1Na1Tq6AUHwESaucmG0qcaBAPUcif8637_p4Zah4WU6s/edit#heading=h.lhu2yo6xjrdn)), if it's not already available on ilifu as a module.

ludwigschwardt, 2024-03-18:
Thanks for the suggestions, Jordan!

When I run your command on the SARAO cluster I get past that point without an error. I find it a bit strange that the code goes straight into `socket.timeout` without the usual retries.

What version of katdal is this, Francesco?

f-carotenuto, 2024-03-19:
Thanks you very much both Jordan and Ludwig for the suggestions!
It was indeed a version problem, as I found out that I was running the 0.20 version (from the katcal_py3 kernel), and everything worked well once I used the 0.21 version from the CASA 6 Kernel in Jupyter. :)
I usually just push the data from SARAO to IDIA, but this time I needed to extract an only a specific field from the whole set of targets observed.

ludwigschwardt, 2024-03-19:
Great, glad to hear that!

closed 2024-03-19 by ludwigschwardt